marcel-licence/ml_synth_organ_example #20: Does RP2040 fail gracefully?

bradanlane, 2023-05-16:
While building the organ sample, I needed a test file and so I used [this midi file](https://onlinesequencer.net/2747142).

I drive the RPiPICO via USB from a midi player running on Windows.

The RPiPICO will crash/lockup once the midi data becomes heavy and relentless. Power cycling is the only option.

TBH: I’m not surprised. I’m more curious if the ML_SynthTools (and/or example code) should fail gracefully by dropping notes or some other behavior vs crashing completely and requiring a reboot.

marcel-licence, 2023-05-16:
I've exported the linked sequence as MIDI file and played it back on the tiny2040. It went through the heavy part but stopped in the next quiet region. I don't think that the organ "core" stopped working. 
I've tested the organ a lot and it was made to work with a single channel input.
A demo midi file is included in the project (in the data folder).
I've recorded another demo to demonstrate some more "stress" for the device.
[brainstorm-20230516200318.zip](https://github.com/marcel-licence/ml_synth_organ_example/files/11490720/brainstorm-20230516200318.zip)

A good test would be to remove the organ from the audio loop and link note on/off to a Serial.printf function. This should allow to check if the problem was produced by the audio part or by some external library (usb midi)

In case of to much computations within the audio loop you will get only some audio glitched. In case some other code keeps the interrupt busy the device may crash.

Hope this helps

bradanlane, 2023-05-16:
I wasn’t overly surprised it locked up. I’m my test, I mapped each instrument to its own channel and had all channels processed by the RP2040. It was not always predictable when it would lock up. The mat consistent was the last 30 seconds. As you noted, there is a lull then the instruments pick up again and then it would lock up.

I anticipate having multiple devices connected with each one handling one or two channels.

closed 2023-05-16 by bradanlane

marcel-licence, 2023-05-16:
I am surprised that it locks up. 
I made a test removing my organ code and just printing messages on NoteOn/Off events. In that case there is a crash:

[image]

I do not know what other messages are transmitted. In that case a USB raw dump would be necessary. I guess something within USB MIDI processing locks up but that is outside of the scope of my project.

marcel-licence, 2023-05-16:
I did another test using my USB to MIDI DIN device to provide the MIDI messages via the RX pin of the Tiny2040. In this test there is no crash. I might just loose some messages (I guess the Rx buffer is to small and the processing cycle to slow)