marcel-licence/ml_synth_organ_example #21: Undefined reference to `ScopeOled_Process()' on ESP32-S3

realny, 2023-08-14:
Hello.

I'm playing with `ml_synth_organ_example` using EPS32-S3 based [ESP32-S3-DEV-KIT-N8R8 board](waveshare.com/wiki/ESP32-S3-DEV-KIT-N8R8) from waveshare.

I've successfuly managed to use MIDI-IN using `6N138` and `PCM5102A` DAC and I have much fun plaing with it. 
However, I have a problem with scope. When I define OLED_OSC_DISP_ENABLED, compilation fails with the following message:

```
undefined reference to `ScopeOled_Process()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
```

I'm using vscode + platformio. My `platfomio.ini` file is:

```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_port = com11
monitor_port = com11
monitor_speed = 115200
lib_deps = 
	adafruit/Adafruit SSD1306@^2.5.7
	adafruit/Adafruit GFX Library@^1.11.7
	https://github.com/marcel-licence/ML_SynthTools.git
	adafruit/Adafruit BusIO@^1.14.1
build_flags =
        -L"${platformio.lib_dir}"
        -lML_SynthTools
```

I'm comiling code using esp32s3 variant of precompiled `SynthTools` library by
placing `ML SynthTools\src\esp32s3\ML_SynthTools.a`  file in `"${platformio.lib_dir}"` folder and renaming it to `libML_SynthTools.a`.

Am I doing something wrong or is it possible, that ESP32-S3 version has no scope support compiled in?

Best regards,
Piotr Poznański

marcel-licence, 2023-08-14:
Hello,

I am happy that you enjoy playing with this project. The scope module is currently missing for ESP32S3. I will try to add it soon. Not sure if there was a reason that it won't work. I will check it soon.

Best regards,
Marcel

marcel-licence, 2023-08-14:
I've added the missing module. After an update it should be able to link the missing function.

realny, 2023-08-14:
Hi.

Thanks for such an immediate response!

You have full right to be happy and proud! :) It's incredible, what you have achieved considering it all works  on just a microcontroller, and not a full blown modern PC!

As for a scope: new version works on my board! Just had to add one line to `ml_synt_organ_example.ino` file :

```cpp
ScopeOled_AddSamples(mono_f, mono_f, SAMPLE_BUFFER_SIZE);
```

In fact, I've made sllight refactor around `Reverb_Process()` invocation. Since both `Reverb_Process()` and `ScopeOled_AddSamples()` needed data in form of `float[]`,  I've extracted it to separate, common block, as follows:

```cpp
#if (defined OLED_OSC_DISP_ENABLED) || (defined OLED_OSC_DISP_ENABLED)
    float mono_f[SAMPLE_BUFFER_SIZE];
    for (int i = 0; i < SAMPLE_BUFFER_SIZE; i++)
    {
        mono_f[i] = mono[i];
    }
#endif
```
Than I can call each functionallity witihin dedicated and independent `#ifdef`-s:

```cpp
#ifdef OLED_OSC_DISP_ENABLED
    ScopeOled_AddSamples(mono_f, mono_f, SAMPLE_BUFFER_SIZE);
#endif
#ifdef REVERB_ENABLED
    Reverb_Process(mono_f, SAMPLE_BUFFER_SIZE); /* post reverb */
    for (int i = 0; i < SAMPLE_BUFFER_SIZE; i++)
    // ... rest of the code
#endif
```

Once again, thank you for providing this new version of library!

Now, when I finally managed to make it all run (and I also included ADC mux code from one of your other projects) I think I could proceed to exploring it further :)

Best regards,
Piotr

marcel-licence, 2023-09-22:
Hi Piotr,
seems I've overlooked your answer. I am confused that the reverb worked because there is one step missing when converting from int32 to float and back (normalizing the samples). But it might work as is as-well. Usually the float values should be limited to -1.0f - 1.0f. But seems this version does work. Does the ESP32S3 use a FPU for multiplications? On other MCUs the reverb might be to slow for execution.

I will push your suggested changes soon to GitHub. Thank you very much for your feedback!

Best regards,
Marcel

closed 2023-09-22 by marcel-licence

marcel-licence, 2023-09-22:
If you like please check https://github.com/marcel-licence/ml_synth_organ_example/tree/OLED_Support