marcel-licence/ml_synth_organ_example #25: Where is Organ_SetLeslCtrl() defined?

Blackaddr, 2024-06-12:
Try to compile the ml_synth_organ_example for Teensy. It's getting a linker failure for a number of missing Organ_* functions, but I think it's correctly using the ML_SynthTools library.

But, when I grep the library and sample repo for the undefined functions, I see their declarations in ml_organ.h but I don't actually see where their definition bodies come from?

undefined reference to `Organ_SetLeslCtrl(unsigned char)
undefined reference to `Organ_SetLeslCtrl(unsigned char)
undefined reference to `Organ_PercussionSet(unsigned char)
undefined reference to `Organ_SetDrawbar(unsigned char, unsigned char)
undefined reference to `Organ_Setup(float)
undefined reference to `Organ_Process_Buf(long*, unsigned char)

Blackaddr, 2024-06-12:
is ML_SynthTools *only* available as a precompiled library?

Blackaddr, 2024-06-12:
This is covered in the discussions. Unfortunately this library is not fully open source and contains close-source components. Disappointing, but closing the issue as this seems to the intent.

marcel-licence, 2024-06-13:
Thanks for the feedback. I will revisit the Teensy project. I've coded a lot of stuff in the last years and decided to part some open parts and some closed parts intentionally. I still maintain the library and will add updates if necessary. In addition to that some functions were very sensitive causing trouble by changing only one line of code breaking the whole optimization. I try to keep the library user friendly. Hope it is okay for you when I come back to this topic when I could check the Teensy compilation.

Blackaddr, 2024-06-13:
@marcel-licence No need to rush. I'm compiling for the Aviate Audio Mutliverse which uses a Teensy Micro under the hood. It has a special effects compiler, but does not allow linking precompiled libraries, it must be compiled from source so it looks like this project won't work for me.

You may want to consider providing source code for "non-optimized" use, at your own risk situations with just naive code implementation. With modern compilers and microcontrollers and embedded CPUs having fast FPUs, honestly many attempts at hand optimization end up slower then just letting the compiler do it.

This would also allow people to do their own optimizations where necessary for new platforms.

marcel-licence, 2024-10-13:
@Blackaddr that is a good point. I think I will try to separate this repository with the open sources and a second one keeping the libraries only. I do not know if some parts could be interesting for commercial use so I think I should be careful before publish all sources.

marcel-licence, 2025-05-17:
I was looking into Aviate Audio Multiverse and it seems like it uses a pretty strict and self-contained workflow. Sometimes these platforms have hidden or undocumented configuration files where custom compile or link flags can be injected, but it’s not clear to me how flexible that environment really is.

closed 2025-05-17 by marcel-licence