marcel-licence/ml_synth_organ_example #28: void Organ_Process_Buf(int32_t *buf, uint8_t len); /* not supported by ESP32, ESP8266 */

icetea06, 2025-04-22:
I'm trying to compile on ESP32 but I get the error message :
ml_synth_organ_example.ino:438:(.text._Z4loopv+0x22): undefined reference to `Organ_Process_Buf(long*, unsigned char)'

In line 94 of ml_organ.h : 
void Organ_Process_Buf(int32_t *buf, uint8_t len); /* not supported by ESP32, ESP8266 */

So I guess Organ_Process_Buf() is not included in the precompiled library for ESP32.
How can I then compile the project ?

icetea06, 2025-04-23:
Thanks for your quick reply !

Yes, I have the latest versions of ML_SynthTools_Lib (2.0.13), ML_SynthTools (2.0.12), and ml_synth_organ_example.
In config.h, I uncommented the line #define BOARD_ESP32_DOIT
My arduino ide version is 2.3.5. My ESP32 boards definitions version is the latest : 3.2.0
I then compile for the ESP32 Dev Module

Here is the compilation log :
>Library ML SynthTools has been declared precompiled:
>Precompiled library in "e:\Documents\Prog (projets)\Arduino\libraries\ML_SynthTools\src\esp32" not found
>Library ML SynthTools Libraries has been declared precompiled:
>Using precompiled library in e:\Documents\Prog (projets)\Arduino\libraries\ML_SynthTools_Lib\src\esp32
>C:/Users/sywin/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2411/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\sywin\AppData\Local\arduino\sketches\310F7DB7EC09A57AF40C6BFB807481FF\sketch\ml_synth_organ_example.ino.cpp.o:(.literal._Z4loopv+0xc): undefined reference to `Organ_Process_Buf(long*, unsigned char)'
>C:/Users/sywin/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2411/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\sywin\AppData\Local\arduino\sketches\310F7DB7EC09A57AF40C6BFB807481FF\sketch\ml_synth_organ_example.ino.cpp.o: in function `loop()':
>E:\Documents\Prog (projets)\Arduino\ml_synth_organ_example/ml_synth_organ_example.ino:438:(.text._Z4loopv+0x1a): undefined reference to `Organ_Process_Buf(long*, unsigned char)'
>collect2.exe: error: ld returned 1 exit status
>exit status 1
>
>Compilation error: exit status 1

Any clue ? I am very impressed by your work and I would very much like to be able to use it.

marcel-licence, 2025-04-23:
The ESP32 board 3.2.0 might not work. The latest 2.x version is the latest working version. They changed something within the ESP32 board breaking some functions.

icetea06, 2025-04-24:
You're right ! It's working with 2.0.17. Thanks for your time !

FYI I had also an issue with line 559 in midi_interface.h : 
pinMode(MIDI_RX2_PIN, INPUT_PULLUP); /* can be connected to open collector output */

I have an FTDI USB to serial adapter connected to my esp32 RX2 pin but the esp32 wasn't receiving anything, so I commented out line 559 and now all is fine !

marcel-licence, 2025-05-17:
Thank you for your feedback. I am happy to hear that it is working.
Please feel free to try out the organ pro demo as-well.
It is strange that the pull-up option causes problems. I will try to keep that in mind.
Do you have a change to measure the electrical signal of the RX2 pin maybe with a scope?

closed 2025-05-17 by marcel-licence